koide3/hdl_localization #1: difference between conventional ndt and this package

weisongwen, 2018-03-19:
hi @koide3  

thanks for your kind effort on this package, could you tell me what is the difference between this package and conventional ndt ?

Best 
Welson

koide3, 2018-03-21:
Hi,
The major difference is that this package provides a capability of UKF-based sensor pose estimation. It allows us to start NDT with a good initial guess and makes the scan matching fast and robust.

weisongwen, 2018-03-21:
Hi, @koide3  Thanks for your kind reply

koide3, 2018-03-22:
Hi,
No, at this time, we've not integrated GPS info with localization. We use GPS for only the mapping phase.

weisongwen, 2018-03-22:
Hi, @koide3 
thanks for your kind reply, in the mapping step. can this package can do real time mapping?
Thanks
Best 
Welson

koide3, 2018-03-22:
Hi,
Yes. We confirmed that this package can run real-time on a standard desktop PC. We expect that it can run on a laptop PC as well, but it depends on the setting.

weisongwen, 2018-03-22:
Hi @koide3 
Thanks for your kind reply, I mean: if the mapping package "hdl_graph_slam" can run real time?
Thanks 
Best 
Welson

koide3, 2018-03-22:
Hi,
Yes, I meant "hdl_graph_slam" can run real-time.

weisongwen, 2018-03-23:
hi @koide3  
thanks for your kind reply, 
best 
welson

JSRGFJZ, 2018-04-02:
@weisongwen 
Hi, did you run the bag? I didn't not get the final result as in the viseos.

weisongwen, 2018-04-02:
Hi， @JSRGFJZ 
I have not test the localization package yet,
Best 
Welson

koide3, 2018-04-02:
Hi, @JSRGFJZ
How did it fail? Did the programs start correctly?

JSRGFJZ, 2018-04-03:
@koide3 
Hi, I notice that when I didn't using IMU to predict, the error is very large.WHY?
is it because of only the lidar is one line instead of multi-line? if I have velodyne-16, could it be better?

weisongwen, 2018-04-03:
Hi @JSRGFJZ  
I think that you may not give a good initialization when you begin the mapping based localization process.

koide3, 2018-04-03:
Hi @JSRGFJZ ,
You are trying to run the bag (hdl_400.bag, 32 line LIDAR), right? I guess the computation power of your PC is insufficient for the default localization parameters. Can you change  *ndt_neighbor_search_method* in hdl_localization.launch to "DIRECT1" and re-try to run the bag? It makes the localization faster significantly.

JSRGFJZ, 2018-04-04:
@koide3 
Actually I already set it to DIRECT1, but the result is not good.

koide3, 2018-04-10:
@JSRGFJZ 
Sorry for the late response. I still can't understand how it fails. Can you try it with a slower playback speed? Like: 
```bash
rosbag play --clock -r 0.1 hdl_400.bag
```

JSRGFJZ, 2018-04-13:
@koide3 
I already tried that, it still doesn't work.
I didn't use IMU, this should be the reason. but I read your thesis, you said it also could, so should I alter some parameters?

koide3, 2018-04-15:
@JSRGFJZ 
Yes, it should be able to use it without IMU. But, the lack of IMU makes the localization weak against quick motions. Make me sure that it works with IMU, right?

JSRGFJZ, 2018-04-16:
Yes, it works with IMU.
I don't know if it is the only situation with me, but I really appreciate your work! Thank you.

j.z.feng@foxmail.com
 
From: koide3
Date: 2018-04-15 22:04
To: koide3/hdl_localization
CC: Jean; Mention
Subject: Re: [koide3/hdl_localization] difference between conventional ndt and this package (#1)
@JSRGFJZ
Yes, it should be able to use it without IMU. But, the lack of IMU makes the localization weak against quick motions. Make me sure that it works with IMU, right?

closed 2018-04-24 by koide3