koide3/hdl_localization #7: Resources for UKF

CtfChan, 2018-11-15:
Hi Kiode3, thanks for this wonderful ROS package. I am a student looking to learn more about the fundamentals of the UKF you implemented. Is there a particular paper that details your implementation?

koide3, 2018-11-15:
Hi @CtfChan ,
If you meant the UKF itself, the following documents would be helpful:
https://www.seas.harvard.edu/courses/cs281/papers/unscented.pdf
https://www.cse.sc.edu/~terejanu/files/tutorialUKF.pdf
https://en.wikipedia.org/wiki/Kalman_filter#Unscented_Kalman_filter

If you are interested in the UKF-based pose estimation, please send a mail to koide@dei.unipd.it from your official mail address. Since the paper is still under review, we have not made it public yet...

CtfChan, 2018-11-16:
cheers thanks man :) I'll take a look

closed 2018-11-16 by CtfChan