koide3/hdl_localization #10: localization error

renzhuli, 2019-01-07:
Hello, I compiled all the passes according to your installation requirements, but when positioning, the pose did not move, always fluctuating around the starting point. Have you ever encountered such a problem? Thanks a lot!
[image]
[image]

closed 2019-04-25 by renzhuli

HappySamuel, 2019-05-17:
Hi @renzhuli 

I also have the same problem. How did you solve this issue?

trns1997, 2020-05-14:
Make sure you are publishing time stamps in the header of your pointcloud topic.

wanderingcar, 2021-11-05:
Hi, @renzhuli @HappySamuel 

I've also had the same problem, and did you solve this issue? If did, then how?

Thanks!