koide3/hdl_localization #11: what is invert_imu

HaoyunXu, 2019-02-09:
On default, the invert_imu is enabled but the localization just failed when the car turns. Then I turned it off, and everything works perfectly find. Could you give me a hint of what this invert_imu does?

koide3, 2019-03-15:
Hi @HaoyunXu ,
If it is enabled, all IMU data will be inverted (multiplied by -1). It was made for backward compatibility with a specific sensor. I think I will make it disable by default. Thanks for reporting.

narutojxl, 2021-02-25:
I did a test, it seems that when using [hdl_400.bag.tar.gz](http://www.aisl.cs.tut.ac.jp/databases/hdl_graph_slam/hdl_400.bag.tar.gz) with imu, the imu's acc and gyro in the bag, we should enable `invert_imu_acc` and `invert_imu_gyro`. 
Note: 
imu's orientation is not used.  
Before using imu, we should firstly transform imu's data into laser, if they is not aligned. Am i correct @koide3 :)?

koide3, 2021-03-01:
Yes, the acc and gyro in the bag are inverted, and thus ```invert_imu_acc``` and ```invert_imu_gyro``` need to be enables. ```plot_status.py``` is useful to validate the IMU axes directions that shows prediction errors with/without IMU. If the errors with IMU are smaller than without IMU, your IMU setting should be fine.

[image]

In case the LIDAR and IMU have different coordinate systems, it's better to change ```odom_child_frame_id``` to the imu frame. Then, input point cloud is transformed in the IMU frame, and localization should be done without problems.

narutojxl, 2021-03-01:
Thanks for koide3's detailed reply. :+1:

DCYzhao, 2025-03-11:
[image]
When using IMU for prediction, the translation prediction is not ideal. May I ask what the issue is, and is this normal?