koide3/hdl_localization #14: velodyne error

melhousni, 2019-03-07:
Hello,
When using this package on my own data, I get this error related to the velodyne noedelet manager :

koide3, 2019-03-15:
Hi @melhousni ,
Could you try to launch the nodelets in standalone mode? 

Change:
<node pkg="nodelet" type="nodelet" name="globalmap_server_nodelet" args="load hdl_localization/GlobalmapServerNodelet $(arg nodelet_manager)">

to:
<node pkg="nodelet" type="nodelet" name="globalmap_server_nodelet" args="standalone hdl_localization/GlobalmapServerNodelet $(arg nodelet_manager)">

HappySamuel, 2019-03-20:
Hi @koide3 

What's the different between using load and standalone?

Samuel

koide3, 2019-03-20:
@HappySamuel ,

In the standalone mode, all the modules are started as separated nodes, while in the load model, they are loaded in a single node (nodelet manager) as nodelets. See http://wiki.ros.org/nodelet for details.

Starting nodelets in the standalone mode is sometimes helpful to find the problem.

HappySamuel, 2019-03-21:
Hi @koide3 

Thank you for the explanation.

Samuel

closed 2019-03-29 by melhousni