koide3/hdl_localization #15: Loaded own 3D pointcloud map orientation RPY = 90', 0', 0'

HappySamuel, 2019-03-20:
Hi,

I have recorded my own 3D pointcloud map and wanna to try it out with hdl_localization package. However, the loaded map orientation if off the plane with RPY=90', 0', 0'
How should i correct the orientation of the loaded 3D pointcloud map?

I use ROS package "pointcloud_to_pcd" to save the map.

Best.
Samuel

HappySamuel, 2019-03-20:
Problem solved while waiting for answer...

For .pcd files, just open it with gedit. Then modify the following arguments...
FIELDS y z x intensity
[image]

By modifying the x y z location, can change the orientation of the pointcloud map.

closed 2019-03-20 by HappySamuel

walterchenchn, 2019-06-09:
Hi,

After i modify the .PCD file, and save the file, the file was been corrupted. How can i open the .PCD file with gedit?

HappySamuel, 2019-06-10:
Hi @walterchenchn 

That's weird. Can you backup the .pcd file, and then try it again.

Best,
Samuel