koide3/hdl_localization #18: Problems with

HappySamuel, 2019-04-29:
Hi @koide3 

I faced an error when running the launch file, **hdl_localization.launch** with the given bag file, **hdl_400.bag**. The alligned pointcloud doesn't match the globalmap, and the odom is also wrong. Here i attached the picture:
[image]

I tried with both DIRECT7 and DIRECT1, unfortunately the situations are the same. Is there a way to make it to run like the pictures you show in the README.md?

Best,
Samuel

koide3, 2019-05-27:
Hi @HappySamuel ,
It may be worth trying play the bag file slowly to find the cause. Can you try like "rosbag play --clock -r 0.25"?

crankler, 2019-06-20:
When I launch all these command , and play the bag, hdl_localization.launch break down and I get this error:

>  [velodyne_nodelet_manager-1] process has died [pid 31237, exit code -11, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=velodyne_nodelet_manager __log:=/home/crank/.ros/log/7d38045a-9359-11e9-881e-a0c589a9ca36/velodyne_nodelet_manager-1.log].
log file: /home/crank/.ros/log/7d38045a-9359-11e9-881e-a0c589a9ca36/velodyne_nodelet_manager-1*.log

Can someone get this error again? How to solve this error ?

txshang, 2019-09-19:
@HappySamuel Do you sovel this problem ？

lixiang103, 2020-08-27:
Hello, I'm having the same problem with you, but changing the resolution and downsampling methods doesn't work. My car is still stuck in the initial position, has your problem been solved?

HappySamuel, 2020-08-28:
@txshang , @lixiang103 

I no longer using this localization package, as it does not work. Currently i am following mcl_3dl.

Best,
Samuel