koide3/hdl_localization #21: localization fail at the same spot

ktk1501, 2019-08-12:
Hello. Thanks for the great works!

I am now running hdl_localization using my own data. I made my own PCD using hdl_graph_slam, and have a right format rosbag file.

It has no problem while running on a small pcd map. I found a problem on a bigger map, it fails to localization always on the same spot. See the video in the link (fails around 3:00) : 
https://youtu.be/pI-KB9ZZ8tA

Changed parameters are here:
invert_imu -> true
cool_time_duration -> 0.1

When 'invert_imu' is false, it fails to localization a way before that spot when it is true. So I am using 'invert_imu' as true. 

Additionally, I have a warning message. I don't know is it related to my problem, but I am posting it : 
"[pcl::VoxelGrid::applyFilter] Leaf size is too small for the input dataset. Integer indices would overflow."

Now, I have a question. Do you remember the 'road-flipped' issue of mine? I thought it is related to the IMU data again, so I checked the source code until 'pose_estimator.hpp'  but I couldn't find any error. (I didn't check inside of 'ukf')

Can you tell me how to solve this problem??

ktk1501, 2019-08-13:
I think it is related to the sparse pointcloud map. So it is even get more score to match the road-plane, than to match the others. 

Will it be solved when use a dense pcd map ? If it is, Is there any way to make a more dense pcd map in hdl_graph_slam? What parameters should I change? 

Best.

koide3, 2019-08-14:
Hi @ktk1501 ,
Yes, I think the sparsity of the map may affect the localization. To make the map dense, you can make "min_edge_interval" and "map_cloud_resolution" smaller (increase the density of keyframes and generate the map with smaller downsample resolution).

https://github.com/koide3/hdl_graph_slam/blob/f7a0472bbb0c48be941fbbe5d3d16c8b01b3e3fe/launch/hdl_graph_slam.launch#L87
https://github.com/koide3/hdl_graph_slam/blob/f7a0472bbb0c48be941fbbe5d3d16c8b01b3e3fe/launch/hdl_graph_slam.launch#L126

ktk1501, 2019-08-14:
Hi @koide3 . Thanks for your kind reply. 

I had created the dense map after adjusting some parameters. Then, I ran hdl_localization to localize. But I met another problem.

I am using two rosbag file, I just call it first.bag and second.bag. Localization was works really well on fisrt.bag. The problem I met above was on second.bag. But now, after using dense map, I found localization failure on the first.bag. 

The localization failure is not look like the one before. It travels to another place, but yes it happens on the same spot. 

You can see the video : https://www.youtube.com/watch?v=Oz5kcGG6y3U

What do think the problem is, and it there any known solution??

Best.

ktk1501, 2019-08-16:
I solved this problem by setting "invert_imu" to false. It was my mistake. 

Thanks for you reply. 

Best.

closed 2019-08-16 by ktk1501