koide3/hdl_localization #22: Try KITTI data but encounter problem that leaf size is too small for the input dataset.

Yukina1996, 2019-09-08:
[image]
[image]
I find that the pose estimator doesn't work and the car doesn't move. I don't know if the problem is the leaf size is too small for the input datase.
So I changed the leaf size of the global map from 1 to 10 and also the input point cloud from 0.1 to 1.0, but the problem still remain.

koide3, 2019-09-17:
Hi @Yukina1996 ,
I'm not sure but it's possible that the map is too large for the downsampling method. The problem may be resolved by changing the VlxelGrid in the following line to ApproximateVoxelGrid which does not cause the leaf size problem.

https://github.com/koide3/hdl_localization/blob/7e1dcdd7f0c8f7b042666c2730470ab6aa0a2945/apps/globalmap_server_nodelet.cpp#L58

Yukina1996, 2019-09-17:
@koide3 Thank you for the reply.I confirm it's the problem of too big map and I'll try ApproximateVoxelGrid later.
I still have another question that is the imu data so important that localization always failed if we didn't use the imu data at the same time. Have you ever tried lidar localization alone based on a big map???

koide3, 2019-09-17:
@Yukina1996 ,
Yes, I've used the localization without IMU. When the resolution of NDT is small, it could be sensitive to quick rotation without IMU data. The convergence might be improved by increasing the resolution (up to 10m).