koide3/hdl_localization #24: The problem when launch the node

txshang, 2019-09-19:
When I use "rosparam set use_sim_time true"   "roslaunch hdl_localization hdl_localization.launch"
  it will stop on the this staement "[ INFO] [1568861604.901954411]: Initializing nodelet with 8 worker threads."
  if I use  "rosparam set use_sim_time false"
  It wiil report the error like this
  "[FATAL] [1568860914.615317735]: Failed to load nodelet '/globalmap_server_nodelet` of type `hdl_localization/GlobalmapServerNodelet` to manager `velodyne_nodelet_manager'
   [FATAL] [1568860914.615321512]: Failed to load nodelet '/hdl_localization_nodelet` of type `hdl_localization/HdlLocalizationNodelet` to manager `velodyne_nodelet_manager'"
   I use ros-indigo on ubuntu14.04.

koide3, 2019-09-22:
What is the version of PCL and g2o? If you're using one you built by yourself, I recommend to switch to ones provided by ROS (the ones you can install with apt). It's been reported that self-built libraries cause this kind of problems.

zorosmith, 2019-12-22:
@txshang hello, have you solved this problem? 
Plz let me know.

zorosmith, 2019-12-22:
@koide3 Hello, following your suggestion, I switched the pcl to the one provided by ROS. 
However, I met the same error.
These are my set-up: ubuntu16.04, ros-kinetic, ros-kinetic-pcl(1.7), cmake_3.15, std::c++11.
Any help will be appreciated!