koide3/hdl_localization #25: Failing to track after a few seconds?

wiku30, 2019-09-19:
Hi,
When I run it with the default launch file (hdl_localization.launch) and the hdl_400.bag, it seems to work for the first few frames, but it stopped moving after a while, seeming to have lost the track.
So in the video provided in the readme, did you use the original launch file, or have you modified some settings?
Thanks.

wiku30, 2019-09-19:
BTW, I feel that the concentric rings on the ground might affect the tracking as they are likely to be static relative to the car instead of the map. Could you briefly tell me how you have solved this problem...

txshang, 2019-09-20:
I have met the same question with you.You can try to adjust the downsample to 0.5 and have a try.

wiku30, 2019-09-20:
@txshang 

I tried it and the tracking persists a few more frames, but still lose after that. I wonder if there is any "debug" message in this package like hdl_graph_slam.

wiku30, 2019-09-20:
@txshang
Solved with another way, but thanks anyway

closed 2019-09-20 by wiku30

WilliamWoo45, 2020-07-13:
@wiku30 Hi Zishuo. I'm having the same problem with you. Could you brief me how you managed to solve it?