koide3/hdl_localization #26: Nodlet fails with SIGSEGV segmenation fault on launch

grimmmyshini, 2019-10-07:
Hey! Thanks for the great work! However I am facing a problem, the globalmapserver nodlet fails with a `Double free or corrupt (out)` segmentation fault. I have tried running the given pcd file and my own pcd file but with no luck. I tried debugging the nodelet with Valgrind, and it seems to me that the node is not able to fetch data from the pcd file. What could be the cause of this error? 

Thanks in advance!

koide3, 2019-10-08:
Hi @grimmmyshini ,
I'm not sure, but I suppose it could be a problem with PCL. What is the version of PCL you are using? Did you build it by yourself or install it with apt?

grimmmyshini, 2019-10-08:
I am using PCL 1.8.0, and I installed it with apt. PCL has downwards compatibility, this should not really happen, right? Could it be anything else?

koide3, 2019-10-10:
I'm not sure, but I saw similar problems caused by self-built PCL binaries. If you built PCL by yourself, could you try to re-install PCL via apt-get?

grimmmyshini, 2019-10-11:
Yeah, I will try purging the existing PCL and reinstall it via apt-get. Thanks!

closed 2019-10-11 by grimmmyshini