koide3/hdl_localization #27: [Failed to load nodelet '/globalmap_server_nodelet`] occasionally

wiku30, 2019-10-13:
It happens occasionally when doing localization while playing rosbag files, but sometimes it doesn't happen.
I wonder why...

zorosmith, 2019-12-22:
I alway meet this error 
Failed to load nodelet '/globalmap_server_nodelet` of type `hdl_localization/GlobalmapServerNodelet` to manager `velodyne_nodelet_manager'
Could any one help?
These are my set-up, ubuntu16.04, ros-kinetic, pcl_1.9, cmake_3.15, std::c++14.

koide3, 2019-12-23:
Hi @zorosmith ,
Did you build pcl1.9 yourself? It's been reported that self-built libraries often cause this kind of error. Try to use apt binaries with : ```sudo apt install ros-kinetic-pcl-ros```

zorosmith, 2019-12-26:
Thank you! @koide3 You are right!
At first, I switch to pcl1.9 in CMakeList.txt. However, it doesn't work and meet the same error.
Then I found that some environment variables in ros workspace still pointed to pcl 1.9 .
Finally, I created a new ros workspace and compile your package in the new one. Solved!

wyq0721, 2020-09-03:
Hi @zorosmith, I meet this error, too!
But my set-up: ubuntu18.04, ros-melodic, pcl_1.8 is not same with you.
Now, i still not launch this node by re-install ros-melodic-pcl-ros.
If you have some advices, please let me know.