koide3/hdl_localization #30: Losing Localization. Please help

vimalrajayyappan, 2019-10-29:
Dear friend,

Thanks for the package.
I had tried using your package for localization where I found some drifts in localization even when Im fusing with IMU.

I used DIRECT7 method for registrationa and localization
NDT_OMP

I saved map at 0.01 resolution

when I try to localize i found some warning like
"Leaf size is too small for the input dataset. Integer indices would overflow."

I altered downsample_resolution and checked as well. Couldnt figure.I attached a localization config file screenshot.Please check.
[image]

Kindly help me to gain the accuracy.
Suggest me some steps.

Thanks a lot

koide3, 2019-10-30:
Hi @vimalrajayyappan ,
Maybe your map is too large for voxelgrid filtering. Change VexelGrid in the following line to ApproximateVoxelGrid which can handle smaller resolutions.

https://github.com/koide3/hdl_localization/blob/624a7613d1639fefc0f67934e4f57dd2a8c9878f/apps/globalmap_server_nodelet.cpp#L58

vimalrajayyappan, 2019-10-30:
Thanks a lot Koi. I will give a try.
Following which can you please share some values of params in Mapping&Localization and their insights. So that I can tune and check their optimality.

Another qn pls:
Should initial pose be highly accurate? Once I give the initial pose whether the algorithm tries to match the best position nearby, before I start moving?

koide3, 2019-10-31:
The following guide would be helpful to tune the parameters.
https://github.com/koide3/hdl_graph_slam#the-mapping-result-is-poor-parameter-tuning-guide

The initial pose should be close to the actual pose (in particular, a large rotational error may affect the convergence). Once you give the initial pose, it updates the estimated pose even if the sensor is fixed.

closed 2019-10-31 by vimalrajayyappan

fatmanurgezmiss, 2021-12-13:
> ctual pose (in particular, a large rotational error may affect the convergence). Once you give the initial pose, it updates the estimated pose even if the sensor is fixed.

> The following guide would be helpful to tune the parameters. https://github.com/koide3/hdl_graph_slam#the-mapping-result-is-poor-parameter-tuning-guide
>
> The initial pose should be close to the [...]

Hi,
I have the same issue that called "Leaf size is too small for the input dataset. Integer indices would overflow."  Then I saw yor answer for that. When I try ApproximateVoxelGrid terminal issue gones but localization fail is continues. Can you help me for that?