koide3/hdl_localization #35: More details about IMU

ricber, 2019-12-18:
Hi, could you please explain how you built the IMU data? Did you use the Velodyne embedded sensor or an external one? How did you calculate the orientation? Did you use the common filters from ROS imu_tools or something else? What about the covariance matrices?

Thank you for the help :)

koide3, 2019-12-19:
Hi @ricber ,
I used the IMU embedded in velodyne HDL32e. I integrated IMU and LIDAR data using my own UKF, so I didn't use the ROS common filters. I assumed a constant observation noise covariance for updating UKF. See the following code which performs the IMU LIDAR integration and the paper for details:

https://github.com/koide3/hdl_localization/blob/master/include/hdl_localization/pose_estimator.hpp
https://journals.sagepub.com/doi/full/10.1177/1729881419841532

ricber, 2019-12-22:
Thank you for your answer and congrats for the paper because it's very well-written :) 
Btw, I've seen the constant observation noise covariance numbers that you put in the code. How did you come up with them? Are they just common practice black magic numbers, or did you implemented some fine-tuning method?

koide3, 2019-12-23:
I've tried to calculate these values from the IMU's spec sheet, but they didn't work well. Then, I put the magic numbers which work better. If you need some theoretical ways to calculate the covariance, the following papers would be help.

https://arxiv.org/pdf/1708.03852.pdf
https://github.com/ethz-asl/ethzasl_msf/raw/master/2013_IROS_lynen_modular_sensor_fusion.pdf

ricber, 2020-01-09:
So basically you get rid of the covariance values from the imu_data msgs, right? I've your same setting: a Velodyne HDL 32E and IMU msgs from the Velodyne embedded sensor. However, if I run hdl_localization without IMU prediction all is working fine. When I add the IMU prediction the localization fails from the beginning. The only difference is that my covariances values of the imu_data msgs are all set to 0. Instead I see a -1 in the first position of the orientation_covariance vector of your hdl_400.bga file. Is that -1 important?

Nova555, 2020-10-14:
Hi, I  the same problem as you. DId you solve it?

ricber, 2020-10-15:
No, sorry. If I remember correctly in the end I've just chose for the localization without IMU.

ricber, 2020-11-22:
Hi @Nova555 , do you have the "-1" in the first position of the orientation_covariance vector of your IMU message?