koide3/hdl_localization #37: c++: error: unrecognized command line option ‘-msse’; did you mean ‘-fdse’?

Young532042725, 2020-03-13:
hello,friends. today, i compile this on Xavier which is NVIDIA's product, it is the ARM architecture different from the intel‘s x86_64. when i build it on Xavier, some errors occur!! like these:
c++: error: unrecognized command line option ‘-msse’; did you mean ‘-fdse’?
c++: error: unrecognized command line option ‘-msse2’
c++: error: unrecognized command line option ‘-msse3’
c++: error: unrecognized command line option ‘-msse4’
c++: error: unrecognized command line option ‘-msse4.1’
c++: error: unrecognized command line option ‘-msse4.2’
c++: error: unrecognized command line option ‘-msse’; did you mean ‘-fdse’?
c++: error: unrecognized command line option ‘-msse2’
c++: error: unrecognized command line option ‘-msse3’
c++: error: unrecognized command line option ‘-msse4’
c++: error: unrecognized command line option ‘-msse4.1’
c++: error: unrecognized command line option ‘-msse4.2’

In the CMakeLists.txt is "add_definitions(-std=c++11 -msse -msse2 -msse3 -msse4 -msse4.1 -msse4.2)
set(CMAKE_CXX_FLAGS "-std=c++11 -msse -msse2 -msse3 -msse4 -msse4.1 -msse4.2")".
which is OK on the intel‘s x86_64 architecture. Can you give me some help ? THANK YOU FIRST!!

cyan-color, 2021-12-21:
did you solve this? if you have soled, pease tell me how to do

ZBoIsHere, 2022-05-11:
does u guys solve this problem

THLL245, 2022-05-24:
Anyone able to solve this problem?

Strinkin, 2024-03-03:
i have the same problem, someone can solve it?