koide3/hdl_localization #40: Odometry covariance

ricber, 2020-03-31:
Hi Koide,
Can I ask you why you didn't fill the covariance matrix of the `nav_msgs::Odometry` odometry message? I see that your UKF class exposes a method called `getCov()`. Could I use this method to fill the covariance of the `/odom` topic?

Thank you.

koide3, 2020-04-06:
Hi @ricber ,
Yes, you can retrieve the covariance using ```getCov()```. But, I expect that the estimated covariance doesn't well represent the true uncertainty of the localization result because the UKF doesn't take scan matching errors into account.

ricber, 2020-04-13:
Ok, thank you!
Also, why did you put to zero the linear twist?

koide3, 2020-05-07:
Yeah, of course, we can put the estimated velocity in the odom message. The reason I didn't is that the velocity was not necessary in my application, and I was lazy to check the velocity output coordinate...

closed 2020-05-07 by ricber