koide3/hdl_localization #41: Seems to me the UFK is almost bypassed

bugerry87, 2020-05-07:
Thanks for sharing your awesome project.

However, I wonder the prediction part of PoseSystem that almost copies the past state expect of some portion of the gyro.
Velocity stays 0 since it gets continuously updated by itself.
Position is from the past frame.

Or do I misunderstand?

koide3, 2020-05-07:
Hi @bugerry87 ,
We empirically found that introducing acceleration affects the estimation stability (in our setting with a noisy acc sensor). So, we use only angular velocity to update the state in the prediction step, and thus the velocity does not change during this step. 

More important part is the correction step that updates all the state parameters including position, velocity, bias, etc. by integrating the prediction result and the scan matching result. The velocity is changed during this step, and thus, in the next prediction step, the position will be shifted with this estimated velocity.

bugerry87, 2020-05-07:
Wow! Thanks for this explanation.

closed 2020-05-07 by bugerry87