koide3/hdl_localization #60: Able to do global localization without initial pose?

WilliamWoo45, 2020-12-28:
Hi! Thanks for sharing your great work.

I've successfully implemented the example bag. I'm wondering is `hdl_localization` able to do **global localization** (i.e. without specify the initial pose) tasks? I tried a few of my own rosbags, but `hdl_localization` failed to do the global localization.

I assume global localization to be a quite powerful and useful function to determine the robot pose when it is lost/kidnapped. The accurate initial pose sometimes can be very **difficult** to obtain. 

Thanks and all the best! :)

Best Regards,
William

Arslan-Z, 2020-12-29:
Hi all!
I'm kind of having the same question. During my robot research project, I find that global localization can be an important & necessary functionality. And currently I did not find any open source 3D localization algorithm that have the global localization ability.
Any comment for this? @koide3 

Thanks.

koide3, 2021-01-02:
Hi all, 

Although hdl_localization is not able to do global localization at this moment, I'm interested in implementing some global localization methods (e.g., RANSAC-based one and BBS). As @Arslan-Z pointed out, it's very rarely to see open source global localization algorithms (AFAIK, only cartographer provides global localization, but it's 2D) while it would be useful for the community. If you could report your use scenario, it would be helpful to determine what needs to be implemented.

WilliamWoo45, 2021-01-02:
Dear Koide,

Thanks for your prompt reply!
Currently, it is true that open source global localization algorithms are rare (I think `Cartographer `is capable to do global localization **in both 2D and 3D** according to their descriptions, but I haven't implement Cartographer in 3D). 

I'm looking for 3D global localization algorithms which can solve the robot localization problem in the industrial warehouse environment, where the robot is extremely easy to get lost or mis-localized to wrong places due to the geometric similarity of the environment (i.e. the similar goods shelves, frames, etc...)

I'll greatly appreaciate if you could improve the `hdl_localization` by including some global localization methods (RANSAC-based and BBS, as you mentioned). I assume the `hdl_localization` empowered with global localization methods is able to perform quite good in the industrial warehouse environment.

Thanks and Best Regards,
William

koide3, 2021-01-04:
> Currently, it is true that open source global localization algorithms are rare (I think Cartographer is capable to do global localization in both 2D and 3D according to their descriptions, but I haven't implement Cartographer in 3D).

Yeah, Cartographer do both 2D and 3D mapping, but the global localization (BBS) is performed on limited to 2D.

WilliamWoo45, 2021-01-06:
Hi Koide,

Thanks a lot for your detailed clarification.
I'm very looking forward to your implementation and I'll stay tuned for it.

Thanks and all the best,
William

Arslan-Z, 2021-01-06:
Thanks for your explanation! @koide3
Definitely I'll also follow up and stay tuned :)

yutouwd, 2021-01-09:
Hi all,

The global localization sounds quite an interesting and promising function. I'll keep tuned and hope to see the realization of this in hdl_localization !

koide3, 2021-01-18:
Hi all,

I implemented some global localization methods (based on BBS, RANSAC, and Teaser++) and pushed the updated code to ```global_localization``` branch (you need to clone [hdl_global_localization](https://github.com/koide3/hdl_global_localization) in addition to hdl_localization). To perform global_localization, you can call ```/relocalize``` service, and then the localizer will be reset with the estimated global pose: 

https://drive.google.com/file/d/1BSROYDuI5_HaL_7QRXAPMk3SIBxeNdG-/view?usp=sharing

It's still preliminary implementation, and enhance both functionality and documentation needs to be enhanced. But, I hope it would be a help for you.

WilliamWoo45, 2021-01-18:
Hi Koide,

Thanks a lot for your speedy implementation.
I'll try and see the performance :)

Thanks and all the best,
William

UriyMihailov, 2023-09-19:
Hi @koide3,

Thanks a lot for this project!
I was wondering if you could provide some links to the articles about global optimization methods that you've implemented in your project?