koide3/hdl_localization #61: How to run live test of hdl_localization

radeonwu, 2021-01-01:
Would like to seek advice on what are the necessary settings in order to run hdl_localization on a real car, with live lidar/imu data inputs?

So far, I can run hdl_localization without issue under simulation mode, i.e., replaying rosbag files provided by this repository, and also rosbag recorded from my own platform. 

But when I run it on a real platform, in rviz the calculated car pose always drifts away from real position.  BTW, I have set the rosparam /use_sim_time to false before launching the hdl_localization node. I have some suspect this might be related to time clock setting but couldn't figure it out.

So your advice is greatly appreciated.

koide3, 2021-01-02:
Just to confirm, can you do ```rosparam get use_sim_time``` and check if use_sim_time is surely set to false?

radeonwu, 2021-01-02:
hi, koide3,
Yes, I confirm use_sim_time value is false before launching hdl_localizatoin node.
I further checked through the old bag file data v.s. current live sensor data, and find out within current live data, due to lidar/imu driver updates, the imu frequency is now around 100hz, while in the old bag data, imu rate is at 200hz. I will try to recover the imu rate to 200hz, but meanwhile would like to check any special requirement that hdl_localization needs for IMU? I am using livox horizon lidar, with internal 6-axis imu. This internal imu output is not of good quality, any suggestion to tune noise covariance value for it?

koide3, 2021-01-04:
> I will try to recover the imu rate to 200hz, but meanwhile would like to check any special requirement that hdl_localization needs for IMU? This internal imu output is not of good quality, any suggestion to tune noise covariance value for it?

It requires only that IMU data has acceleration and angular velocity, and no other special requirements required. For noise covariance tuning, [this](https://github.com/ethz-asl/kalibr/wiki/IMU-Noise-Model) would be helpful. But, I expect the difference on the noise covariance doesn't affect the estimation so much.

> I further dig into the live imu data, and find that in every 10 imu msgs, 9 have time interval of 5ms, but 1 has 50ms interval, which leads to around 100hz frequency value when checking with "rostopic hz" command. Could this be the reason that causes localization to fail(can ukf handle this case?)

I think the deviation of the IMU frequency would not be a problem as long as the IMU data has correct timestamps