koide3/hdl_localization #62: How to use an external IMU?

OriWong, 2021-01-08:
Hi koide3, 
Thanks for your great work!
 I am trying to use a velodyne-16 lidar and an external  IMU. Maybe I need to set up the calibration. Should I make it the same as HDL32E  ?

koide3, 2021-01-09:
hdl_localization assumes that the IMU and LIDAR coordinate systems are the same, and thus cannot use an external IMU in the current form. A workaround would be to transform the input cloud before feeding it to hdl_localiation so its coordinate system becomes the same as the IMU coordinate system.

OriWong, 2021-01-10:
> hdl_localization assumes that the IMU and LIDAR coordinate systems are the same, and thus cannot use an external IMU in the current form. A workaround would be to transform the input cloud before feeding it to hdl_localiation so its coordinate system becomes the same as the IMU coordinate system.

THANK YOU! I will try it.

closed 2021-01-11 by OriWong