koide3/hdl_localization #66: Hello, there are some errors, I don't know how to solve.

mengxingshifen1218, 2021-01-26:
```c++
/home/seivl/hdl_ws/src/hdl_localization/apps/hdl_localization_nodelet.cpp: In member function ‘pcl::Registration<pcl::PointXYZI, pcl::PointXYZI>::Ptr hdl_localization::HdlLocalizationNodelet::create_registration() const’:
/home/seivl/hdl_ws/src/hdl_localization/apps/hdl_localization_nodelet.cpp:85:115: error: no matching function for call to ‘ros::NodeHandle::param(const char [27], const char [8]) const’
     std::string ndt_neighbor_search_method = private_nh.param<std::string>("ndt_neighbor_search_method", "DIRECT7");
                                                                                                                   ^
In file included from /opt/ros/kinetic/include/ros/ros.h:45:0,
                 from /home/seivl/hdl_ws/src/hdl_localization/apps/hdl_localization_nodelet.cpp:6:
/opt/ros/kinetic/include/ros/node_handle.h:2120:8: note: candidate: template<class T> bool ros::NodeHandle::param(const string&, T&, const T&) const
   bool param(const std::string& param_name, T& param_val, const T& default_val) const
        ^
/opt/ros/kinetic/include/ros/node_handle.h:2120:8: note:   template argument deduction/substitution failed:
/home/seivl/hdl_ws/src/hdl_localization/apps/hdl_localization_nodelet.cpp:85:115: note:   cannot convert ‘"DIRECT7"’ (type ‘const char [8]’) to type ‘std::__cxx11::basic_string<char>&’
     std::string ndt_neighbor_search_method = private_nh.param<std::string>("ndt_neighbor_search_method", "DIRECT7");
                                                                                                                   ^
In file included from /opt/ros/kinetic/include/ros/ros.h:45:0,
                 from /home/seivl/hdl_ws/src/hdl_localization/apps/hdl_localization_nodelet.cpp:6:
/opt/ros/kinetic/include/ros/node_handle.h:2153:5: note: candidate: T ros::NodeHandle::param(const string&, const T&) [with T = std::__cxx11::basic_string<char>; std::__cxx11::string = std::__cxx11::basic_string<char>] <near match>
   T param(const std::string& param_name, const T& default_val)
     ^
/opt/ros/kinetic/include/ros/node_handle.h:2153:5: note:   passing ‘const ros::NodeHandle*’ as ‘this’ argument discards qualifiers
/home/seivl/hdl_ws/src/hdl_localization/apps/hdl_localization_nodelet.cpp:86:75: error: no matching function for call to ‘ros::NodeHandle::param(const char [15], double) const’
     double ndt_resolution = private_nh.param<double>("ndt_resolution", 1.0);
                                                                           ^
In file included from /opt/ros/kinetic/include/ros/ros.h:45:0,
                 from /home/seivl/hdl_ws/src/hdl_localization/apps/hdl_localization_nodelet.cpp:6:
/opt/ros/kinetic/include/ros/node_handle.h:2120:8: note: candidate: template<class T> bool ros::NodeHandle::param(const string&, T&, const T&) const
   bool param(const std::string& param_name, T& param_val, const T& default_val) const
        ^
/opt/ros/kinetic/include/ros/node_handle.h:2120:8: note:   template argument deduction/substitution failed:
/home/seivl/hdl_ws/src/hdl_localization/apps/hdl_localization_nodelet.cpp:86:75: note:   cannot convert ‘1.0e+0’ (type ‘double’) to type ‘double&’
     double ndt_resolution = private_nh.param<double>("ndt_resolution", 1.0);
                                                                           ^
In file included from /opt/ros/kinetic/include/ros/ros.h:45:0,
                 from /home/seivl/hdl_ws/src/hdl_localization/apps/hdl_localization_nodelet.cpp:6:
/opt/ros/kinetic/include/ros/node_handle.h:2153:5: note: candidate: T ros::NodeHandle::param(const string&, const T&) [with T = double; std::__cxx11::string = std::__cxx11::basic_string<char>] <near match>
   T param(const std::string& param_name, const T& default_val)
     ^
/opt/ros/kinetic/include/ros/node_handle.h:2153:5: note:   passing ‘const ros::NodeHandle*’ as ‘this’ argument discards qualifiers
hdl_localization/CMakeFiles/hdl_localization_nodelet.dir/build.make:62: recipe for target 'hdl_localization/CMakeFiles/hdl_localization_nodelet.dir/apps/hdl_localization_nodelet.cpp.o' failed
make[2]: *** [hdl_localization/CMakeFiles/hdl_localization_nodelet.dir/apps/hdl_localization_nodelet.cpp.o] Error 1
CMakeFiles/Makefile2:4338: recipe for target 'hdl_localization/CMakeFiles/hdl_localization_nodelet.dir/all' failed
make[1]: *** [hdl_localization/CMakeFiles/hdl_localization_nodelet.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4" failed

```

koide3, 2021-01-27:
I suppose the error is caused by the const qualifier of ```create_registration``` method. You can just remove it. Note that we do not support environments earlier than ROS melodic. If it doesn't work, please consider using docker to run it on a virtual melodic environment.

mengxingshifen1218, 2021-01-27:
thank you, I will have a try

Oyssster, 2021-03-12:
> I suppose the error is caused by the const qualifier of `create_registration` method. You can just remove it. Note that we do not support environments earlier than ROS melodic. If it doesn't work, please consider using docker to run it on a virtual melodic environment.

Hello koide3, thank you so much for being able to open source such excellent work. But I have a question that can the program run on kinetic?