koide3/hdl_localization #67: Hdl_localization map-->odom--> base link

Magmanat, 2021-02-05:
Hello, i am setting up navigation on my robot and i would like to use move_base which needs a map --> odom --> base link transformation where map and odom are static frames
Map being the global map frame which never changes and odom which is the local reference frame which changes over time to help with drift issues

Right now hdl_localization only provides a map --> base_link transformation which is the direct odom transformation which i am unable to use, do you know any way to use move_base with hdl localization

koide3, 2021-02-06:
Recently, I updated the code so that it issues map -> odom transformation if odom -> base_link exists. Please try the latest commit.

Magmanat, 2021-02-07:
Currently the map --> base link transformation uses the /odom topic published by hdl_localization to do this transformation. So if instead i provide a odom --> base link transformation using the /odom topic hdl_localization will then provide a map --> odom transformation ? 

Thanks for the reply, if this is the case i will try the latest commit

koide3, 2021-02-07:
Yes, if you provide odom -> base_link transform, hdl_localization automatically detect it and publishes map -> odom transform to maintain a proper tree

Magmanat, 2021-02-08:
Thanks for all your cool packages koide3 and the quick explanation, it works thank you

closed 2021-02-08 by Magmanat

kunnalparihar, 2021-07-10:
Hello, I want to use wheel odometry data in this package. So if I publish odom -> base_link transform using wheel odometry.
Is this the correct implementation?

kunnalparihar, 2021-07-12:
This worked for me.