koide3/hdl_localization #68: can not use hdl_global_localization to init hdl

narutojxl, 2021-02-26:
Hi @koide3,
Thanks for your hard wok. I run hdl with `use_global_localization = true, specify_init_pose = false`, then in the terminal `rosservice call /relocalize`,  hdl can not be initialized with hdl_global_localization. 
I modifed [hdl_localization_nodelet.cpp](https://github.com/narutojxl/hdl_localization_noted/blob/develop/apps/hdl_localization_nodelet.cpp)  3 places in code to let hdl init successfuly and test OK.  Because i comment  the raw code with chinese, when trying to give a PR, it will include some trival differences, so i point out 3 places in here.  Hope to be useful :).  
* [init stage call "/relocalize"](https://github.com/narutojxl/hdl_localization_noted/blob/develop/apps/hdl_localization_nodelet.cpp#L82-L93)  
* [set first laser in laser callback](https://github.com/narutojxl/hdl_localization_noted/blob/ab5f019b69c05ca9cbebe1247d8fbefcf78d808a/apps/hdl_localization_nodelet.cpp#L206-L228)
* [capture mutex when we enter "relocalize" callback](https://github.com/narutojxl/hdl_localization_noted/blob/ab5f019b69c05ca9cbebe1247d8fbefcf78d808a/apps/hdl_localization_nodelet.cpp#L372)

koide3, 2021-02-28:
Thanks! I'll update the code soon according to your suggestion 👍

narutojxl, 2021-03-01:
You are welcome :)

Jed-t, 2021-03-12:
hello，when i compile this project on ubuntu16.04&kinetic，met some errors
[https://github.com/koide3/hdl_localization/issues/71](url)
can you help me, does it need to be compiled on 20.04?