koide3/hdl_localization #69: cmake error

whuzs, 2021-03-11:
when i using catkin_make -DCMAKE_BUILD_TYPE=Release -DBUILD_VGICP_CUDA=ON, it occours:

CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
CUDA_cublas_device_LIBRARY (ADVANCED)
    linked by target "fast_vgicp_cuda" in directory /home/zhangshun/zs_work/hdl_localization/src/fast_gicp

why? thank you

mericdurukan, 2021-03-11:
You should update your cmake to 3.13.3.

Oyssster, 2021-03-12:
> You should update your cmake to 3.13.3.

Hi,@mericdurukan .
`no matching function for call to ‘transformPointCloud(std::__cxx11::string&, pcl::PointCloud<pcl::PointXYZI>&, pcl::PointCloud<pcl::PointXYZI>&, tf2_ros::Buffer&)’
     if(!pcl_ros::transformPointCloud(odom_child_frame_id, *pcl_cloud, *cloud, this->tf_buffer)) `
when I run catkin_make to compile the file: hdl_localization_nodelet.cpp, I met this error. Have you compile the code successfully? I compile the code on ubuntu16.04 ,ros-kinetic and using cmake 3.17.5.

whuzs, 2021-03-12:
> You should update your cmake to 3.13.3.

thanks

Oyssster, 2021-03-13:
> > You should update your cmake to 3.13.3.
> 
> thanks

Hi @whuzs , do you compile the package on ubuntu18.04 ros-melodic? I encounter the error while compile the packages.

mericdurukan, 2021-03-13:
> > > You should update your cmake to 3.13.3.
> > 
> > 
> > thanks
> 
> Hi @whuzs , do you compile the package on ubuntu18.04 ros-melodic? I encounter the error while compile the packages.

Hi @Oyssster,

 I compiled the package on ubuntu 18.04 with ros-melodic without any errors

whuzs, 2021-03-13:
> > > You should update your cmake to 3.13.3.
> > 
> > 
> > thanks
> 
> Hi @whuzs , do you compile the package on ubuntu18.04 ros-melodic? I encounter the error while compile the packages.

yes, my environment is ubuntu18.04 and ros-melodic, in the beginning, i can compile the packages in cpu, but it failed in gpu, my cuda version is 10.0, it occurs the above error, it is because of cmake version is too older( install ros, cmake version is 3.10), later, i update my cmake version to 3.18, it success.

DongWan0285, 2021-03-25:
Hi, this problem is sloved it?

I have same problem 

I compiled the package on ubuntu 16.04 with ros-kinetic and cmake 1.13.3 
pcl_ros, pcl 1.7.2