koide3/hdl_localization #71: catkin_make error

Jed-t, 2021-03-12:
/hdl_localization/apps/hdl_localization_nodelet.cpp:529:8: error: ‘atomic_bool’ in namespace ‘std’ does not name a type
   std::atomic_bool relocalizing;

when I run catkin_make I met this error

Oyssster, 2021-03-12:
`no matching function for call to ‘transformPointCloud(std::__cxx11::string&, pcl::PointCloud<pcl::PointXYZI>&, pcl::PointCloud<pcl::PointXYZI>&, tf2_ros::Buffer&)’
     if(!pcl_ros::transformPointCloud(odom_child_frame_id, *pcl_cloud, *cloud, this->tf_buffer)) `
when I run catkin_make I met this error. Is there somebody can tell me how to solve the problem?

Oyssster, 2021-03-12:
> /hdl_localization/apps/hdl_localization_nodelet.cpp:529:8: error: ‘atomic_bool’ in namespace ‘std’ does not name a type
> std::atomic_bool relocalizing;
> 
> when I run catkin_make I met this error

Hi Jed-t, you can try to add the follow code in the CMakeList.txt in the hdl_localization package.
`set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++14")`

Jed-t, 2021-03-12:
> `no matching function for call to ‘transformPointCloud(std::__cxx11::string&, pcl::PointCloud<pcl::PointXYZI>&, pcl::PointCloud<pcl::PointXYZI>&, tf2_ros::Buffer&)’ if(!pcl_ros::transformPointCloud(odom_child_frame_id, *pcl_cloud, *cloud, this->tf_buffer)) `
> when I run catkin_make to compile the file: hdl_localization_nodelet.cpp, I met this error. Is there somebody can tell me how to solve the problem?

Jed-t, 2021-03-12:
> `no matching function for call to ‘transformPointCloud(std::__cxx11::string&, pcl::PointCloud<pcl::PointXYZI>&, pcl::PointCloud<pcl::PointXYZI>&, tf2_ros::Buffer&)’ [...]

i have tried before,and i also met the same problem as yours,,,,,,don't know how to solve it.

Oyssster, 2021-03-12:
> i have tried before,and i also met the same problem as yours,,,,,,don't know how to solve it.

@Jed-t Do you compile in ubuntu16.04 and ros-kinetic?

Jed-t, 2021-03-12:
> @Jed-t Do you compile in ubuntu16.04 and ros-kinetic?

yes，does it work on 20.04？

Oyssster, 2021-03-12:
> does it work on 20.04？

I'm don't know whether it works on 20.04. I use 16.04.[doge]

Jed-t, 2021-03-12:
sad

Jed-t, 2021-03-12:
> > does it work on 20.04？
> 
> I'm don't know whether it works on 20.04. I use 16.04.[doge]

i have tried the earlier version of this project, the imu seems does not work, and the author update a new version,then i cannot compile .sad,,,

narutojxl, 2021-03-13:
@Jed-t,  hope [this help](https://github.com/narutojxl/hdl_global_localization_noted/issues/1#issuecomment-797865491).

Jed-t, 2021-03-13:
> @Jed-t, hope [this help](https://github.com/narutojxl/hdl_global_localization_noted/issues/1#issuecomment-797865491).

ok,thanks a lot !