koide3/hdl_localization #74: ERROR: Service [/relocalize] is not available.

zxl19, 2021-04-14:
Hi @koide3 !

I use hdl_localization to run my own code. However, localization failed after several frames. I tried `/relocalize` service but encountered the following error:

```bash
rosservice list
```

```bash
/globalmap_server_nodelet/get_loggers
/globalmap_server_nodelet/set_logger_level
/hdl_global_localization/get_loggers
/hdl_global_localization/query
/hdl_global_localization/set_engine
/hdl_global_localization/set_global_map
/hdl_global_localization/set_logger_level
/hdl_localization_nodelet/get_loggers
/hdl_localization_nodelet/set_logger_level
/play_1618372696267675442/get_loggers
/play_1618372696267675442/pause_playback
/play_1618372696267675442/set_logger_level
/rosout/get_loggers
/rosout/set_logger_level
/rviz_1618372690694040347/get_loggers
/rviz_1618372690694040347/load_config
/rviz_1618372690694040347/reload_shaders
/rviz_1618372690694040347/save_config
/rviz_1618372690694040347/set_logger_level
/velodyne_nodelet_manager/get_loggers
/velodyne_nodelet_manager/list
/velodyne_nodelet_manager/load_nodelet
/velodyne_nodelet_manager/set_logger_level
/velodyne_nodelet_manager/unload_nodelet
```

```bash
source ./devel/setup.bash
rosservice call /relocalize
```

```bash
ERROR: Service [/relocalize] is not available.
```

JaredLee1218, 2021-04-15:
rosbag play --clock XXX.bag gogo

zxl19, 2021-04-21:
Hi @JaredLee1218 ! I did use `rosbag play --clock` when playing the bag, but somehow I can't call the `/relocalize` service :(

closed 2022-04-15 by zxl19

sangsongzhen, 2024-06-06:
i met the same problem ,