koide3/hdl_localization #81: Using wheel odometry data

kunnalparihar, 2021-07-10:
Hello, I want to use wheel odometry data in this package. Can you please guide me how to add wheel odometry data (x,y,yaw)?

Thanks in advance

koide3, 2021-07-11:
To use wheel odometry data, you need to publish the transformation between ```odom``` and ```sensor``` frames on TF. Then, set ```enable_robot_odometry_prediction``` to ```true```.

https://github.com/koide3/hdl_localization/blob/master/launch/hdl_localization.launch

kunnalparihar, 2021-07-12:
Thanks. It worked for me. 
The overall trajectory looks good when compared with the GPS trajectory, but at some places, the localization trajectory is not smooth (Has some zig zag). How can improve the smoothness of my trajectory? Which parameter needs to be tweaked?
[image]

nyxrobotics, 2022-11-28:
I set enable_robot_odometry_prediction to true but localization does not follow odom. How can I check if it works?