koide3/hdl_localization #88: NDT_CUDA crash

narutojxl, 2021-12-15:
Hi author,
I build fast_gicp with `option(BUILD_VGICP_CUDA "Build GPU-powered VGICP" ON)`. Set `reg_method` is `NDT_CUDA_D2D` or `NDT_CUDA_P2D`, hdl  will crash when received laser points. But when using with [`FastVGICP_CUDA` and `FastVGICP(disable BUILD_VGICP_CUDA)`](https://github.com/narutojxl/hdl_localization_noted/blob/0d8489312e051b578273a9b57fcf7f8c8e6c59aa/apps/hdl_localization_nodelet.cpp?_pjax=%23js-repo-pjax-container%2C%20div%5Bitemtype%3D%22http%3A%2F%2Fschema.org%2FSoftwareSourceCode%22%5D%20main%2C%20%5Bdata-pjax-container%5D#L158-L219) , there is no problem. It seems that it is  `fast_gicp::NDTCuda` lib problem. My OS is ubuntu20.04, noetic, PCL is 1.10. When debug with gdb, there is no backtrace, so i only have the terminal log. Thanks for your time very much :)
>
	nodelet: /build/pcl-gWGA5r/pcl-1.10.0+dfsg/kdtree/include/pcl/kdtree/impl/kdtree_flann.hpp:138: int pcl::KdTreeFLANN<PointT, Dist>::nearestKSearch(const PointT&, int, std::vector<int>&, std::vector<float>&) const [with PointT = pcl::PointXYZI; Dist = flann::L2_Simple<float>]: Assertion `point_representation_->isValid (point) && "Invalid (NaN, Inf) point coordinates given to nearestKSearch!"' failed.
	[velodyne_nodelet_manager-1] process has died [pid 279759, exit code -6, cmd /opt/ros/noetic/lib/nodelet/nodelet manager __name:=velodyne_nodelet_manager __log:=/home/jxl/.ros/log/6f74d344-5d6d-11ec-9054-af2fc89854e3/velodyne_nodelet_manager-1.log].
	log file: /home/jxl/.ros/log/6f74d344-5d6d-11ec-9054-af2fc89854e3/velodyne_nodelet_manager-1*.log

rzhao88, 2021-12-22:
I ran into this issue as well but it was crashing on FastVGICP_CUDA as well. I was able to resolve this by going to commit fa23f9b493347452ea3a029b088b7694f766053f. It seems the Eigen upgrade broke the CUDA stuff.

narutojxl, 2021-12-30:
Hi @rzhao88, sorry for late reply. I tried to switch to  `fa23f9b493347452ea3a029b088b7694f766053f` commit of fast_gicp. When using with `NDT_CUDA_D2D` and `NDT_CUDA_P2D`, hdl complains tf error. But when using with `FastVGICP_CUDA`, hdl is normal. 
> 
	Error:   TF_NAN_INPUT: Ignoring transform for child_frame_id "base_footprint" from authority "unknown_publisher" because of a nan value in the transform (nan nan nan) (0.000000 0.000000 0.000000 1.000000)
		 at line 240 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp
	[ERROR] [1640833458.491322637, 53.560000000]: Ignoring transform for child_frame_id "base_footprint" from authority "unknown_publisher" because of a nan value in the transform (nan nan nan) (0.000000 0.000000 0.000000 1.000000)

gaetan-falcand, 2022-08-09:
Hi,
I tried to use `NDT_CUDA_D2D` and `NDT_CUDA_P2D`, and I got the same error than @narutojxl, is there any upcomming update to fix this @koide3 ? Thank you for the answer.

RaviBeagle, 2023-01-02:
I get the same error. Is the update to Eigen that has to be done ?

ceyhunblr, 2023-01-27:
Anyone have a complete solution?

Yu1107, 2023-08-11:
sudo apt purge libpcl-dev

git clone https://github.com/ros-perception/perception_pcl.git