koide3/hdl_localization #92: Hi, My Localization is not work

Minsu-Kim-KNU, 2022-06-17:
Hi, I'm try to use this package with VLP16.
I already created my global map with LIO-SAM, and I need to perform localization. 
But, when I launch this package, Odometry is not moving. 
Below picture is my launch file with my specific topics, if anyone has a solution, please feel free to reply
Thakns.
[image]

Minsu-Kim-KNU, 2022-06-20:
also, 
[image]

this error is keep occured. what is the problem?

cyhunblr, 2023-04-07:
Could you share your tf tree?