koide3/hdl_localization #96: ROS2 Support

grassjelly, 2022-09-25:
HI, just curious will ROS2 support be part of your roadmap? Thank you!

koide3, 2022-09-25:
I will not add a ROS2 support for this package because I'm planning to release a new localization package with ROS1/ROS2 support.

grassjelly, 2022-09-25:
Thanks for your reply. Do you have a rough estimate of its release date? Looking forward. Thank you

koide3, 2022-09-26:
I expect it would be around Jan ~ Mar in 2023 (can largely vary).

grassjelly, 2022-09-26:
Got it. Thank you!

closed 2022-09-27 by grassjelly

Alex-Beh, 2022-11-08:
@grassjelly Maybe you can check this [repo](https://github.com/DataspeedInc/hdl_localization/tree/ros2) for ROS2 Implementation of hdl_localization.

@koide3 May I know the difference between hdl_localization & the new coming localization package? Do you have any preprint paper for that?

grassjelly, 2022-11-08:
Thanks @Alex-Beh. Have you tried this ROS2 port yet?

koide3, 2022-11-08:
> @koide3 May I know the difference between hdl_localization & the new coming localization package? Do you have any preprint paper for that?

I wrote it from scratch with the tight coupling LiDAR and IMU costs that make it extremely robust to sensor motion and degeneration. I will submit it to some conference but the manuscript has not been available yet. Stay tuned!

grassjelly, 2022-11-08:
@koide3 that sounds really promising. Looking forward to the release. Do you happen to know more 3D localization algorithm that supports ROS2?

pyc5714, 2023-05-21:
@grassjelly @koide3   Did you implement the hdl_localization using this [repo](https://github.com/DataspeedInc/hdl_localization/tree/ros2) in ROS2??

I'm trying to get that going, but I haven't running yet because there seems to be some errrors I need to fix with the dependecy and launch file..