koide3/hdl_localization #98: Topic names are hardcoded

RaviBeagle, 2022-11-05:
Hello developers,

The topic names here are hard coded:

```
    if (use_imu) {
      NODELET_INFO("enable imu-based prediction");
      imu_sub = mt_nh.subscribe("/gpsimu_driver/imu_data", 256, &HdlLocalizationNodelet::imu_callback, this);
    }
    points_sub = mt_nh.subscribe("/velodyne_points", 5, &HdlLocalizationNodelet::points_callback, this);
```

So inspite of the launch file:

```
  <arg name="points_topic" default="/velodyne_points" />

...
      <remap from="/carla/ego_vehicle/lidar" to="$(arg points_topic)"/>
      <remap from="/gpsimu_driver/imu_data" to="$(arg imu_topic)" />

```
Has no effect

RaviBeagle, 2023-01-02:
I dont see any effect of using the remap tag in the launch file. Dont know if this is a ROS bug or not ?

koide3, 2023-01-04:
The remapping is wrong. You should do ```<remap from="/velodyne_points" to="/your/lidar/topic"/>```

rreddy78, 2023-01-13:
Thanks a lot. With this change and a few others I am able to run hdl_graph_slam and hdl_localization with CARLA simulation generated data.

RaviBeagle, 2023-01-13:
Thanks a lot. With this change and a few others I am able to run hdl_graph_slam and hdl_localization with CARLA simulation generated data.

closed 2023-01-13 by RaviBeagle