koide3/hdl_localization #113: Some prombles with catkin_make

DAR-fly, 2023-04-08:
I want to run this code with ubuntu20.04,it run with following errors:
`/usr/bin/ld: CMakeFiles/gicp_align.dir/src/align.cpp.o:(.rodata._ZTVN3pcl6search6SearchINS_8PointXYZEEE[_ZTVN3pcl6search6SearchINS_8PointXYZEEE]+0x20): undefined reference to `pcl::search::Search<pcl::PointXYZ>::getName() const'
/usr/bin/ld: CMakeFiles/gicp_align.dir/src/align.cpp.o:(.rodata._ZTVN3pcl6search6KdTreeINS_8PointXYZENS_11KdTreeFLANNIS2_N5flann9L2_SimpleIfEEEEEE[_ZTVN3pcl6search6KdTreeINS_8PointXYZENS_11KdTreeFLANNIS2_N5flann9L2_SimpleIfEEEEEE]+0x20): undefined reference to `pcl::search::Search<pcl::PointXYZ>::getName() const'
/usr/bin/ld: /home/clf/hdl_global_localization/devel/lib/libfast_gicp.so: undefined reference to `pcl::search::Search<pcl::PointNormal>::getName() const'
/usr/bin/ld: /home/clf/hdl_global_localization/devel/lib/libfast_gicp.so: undefined reference to `pcl::search::Search<pcl::PointXYZI>::getName() const'
collect2: error: ld returned 1 exit status
make[2]: *** [fast_gicp/CMakeFiles/gicp_align.dir/build.make:190：/home/clf/hdl_global_localization/devel/lib/fast_gicp/gicp_align] 错误 1
make[1]: *** [CMakeFiles/Makefile2:612：fast_gicp/CMakeFiles/gicp_align.dir/all] 错误 2
make[1]: *** 正在等待未完成的任务....
/usr/bin/ld: /home/clf/hdl_global_localization/devel/lib/libfast_gicp.so: undefined reference to `pcl::search::Search<pcl::PointXYZ>::getName() const'
/usr/bin/ld: /home/clf/hdl_global_localization/devel/lib/libfast_gicp.so: undefined reference to `pcl::search::Search<pcl::PointNormal>::getName() const'
/usr/bin/ld: /home/clf/hdl_global_localization/devel/lib/libfast_gicp.so: undefined reference to `pcl::search::Search<pcl::PointXYZI>::getName() const'
collect2: error: ld returned 1 exit status
make[2]: *** [fast_gicp/CMakeFiles/gicp_kitti.dir/build.make:190：/home/clf/hdl_global_localization/devel/lib/fast_gicp/gicp_kitti] 错误 1
make[1]: *** [CMakeFiles/Makefile2:585：fast_gicp/CMakeFiles/gicp_kitti.dir/all] 错误 2
`
What problem is causing this？From my point of view,my version of pcl is 1.10,but code requires 1.7,is it?
Thanks a lot!

closed 2023-04-11 by DAR-fly

qimao7213, 2023-06-01:
hi, i meet the same problem. Have you fixed it ?

qimao7213, 2023-06-01:
hi, i just fixed this problem. You can have a try.  
`find_package(catkin REQUIRED COMPONENTS
  roscpp
  # pcl_ros
)`
comment "pcl_ros".