koide3/hdl_localization #128: Available in ROS2?

shavargo109, 2024-05-21:
Hi, I have been using this code for localization for a year in ROS1 environment, and i want to change my robot system using ROS2. So, 

1. Is it any support for hdl_localization to run in ROS2? 
2. Or it is suggested to run the localization in ROS1 environment and bridge the topic to ROS2?
Any helps would be appreciated, thanks!

schuethedennis, 2024-07-02:
Hi, is there anything going on in this direction? I would also like to use it and currently working on a ROS2 version, but when there is a solution already I don't have to do the work.

woshinidiesb, 2025-07-28:
> Hi, is there anything going on in this direction? I would also like to use it and currently working on a ROS2 version, but when there is a solution already I don't have to do the work.嗨，这个方向有什么事吗？我也想使用它，目前正在使用ROS2版本，但当已经有解决方案时，我不必做这项工作。

hi,whether you finished this work that about translate the version?