koide3/hdl_localization #130: How to configure odom frame for move_base with only LiDAR and IMU odometry (no wheel odometry)?

emirhancibir, 2024-12-03:
I am using the hdl_localization package with move_base, but I don't have wheel odometry. My robot relies solely on LiDAR and IMU odometry for navigation.

Here is my current setup:

The hdl_localization node publishes the transform between the odom frame and my sensor frame (laser) via the /odom topic.
I am using a static transform tree:
laser -> base_link -> ...
However, move_base requires an odom frame. I am unsure how to configure this in my current setup.

Should I use the odom frame directly from hdl_localization, or is there a better approach for integrating the odometry output with move_base?
How do I ensure the odom frame is consistent with move_base requirements in a system with no wheel odometry?
Any guidance or best practices would be greatly appreciated.

  
  
  <arg name="nodelet_manager" default="velodyne_nodelet_manager" />
  <arg name="points_topic" default="cloud_all_fields_fullframe" />
  
  
  <arg name="odom_child_frame_id" default="laser" />

  
  <arg name="use_imu" default="true" />
  <arg name="invert_imu_acc" default="false" />
  <arg name="invert_imu_gyro" default="false" />
  <arg name="use_global_localization" default="true" />
  <arg name="imu_topic" default="multiScan/imu" />
  <arg name="enable_robot_odometry_prediction" value="false" />
  <arg name="robot_odom_frame_id" value="odom" />
  <arg name="plot_estimation_errors" value="false" />

  <include file="$(find hdl_global_localization)/launch/hdl_global_localization.launch" if="$(arg use_global_localization)" />

  
  <node pkg="nodelet" type="nodelet" name="$(arg nodelet_manager)" args="manager" output="screen"/>

    
    <node pkg="nodelet" type="nodelet" name="globalmap_server_nodelet" args="load hdl_localization/GlobalmapServerNodelet $(arg nodelet_manager)"  output="screen">
      <param name="globalmap_pcd" value="$(find servant_t)/pcd/liorf2/GlobalMap.pcd" />
      <param name="convert_utm_to_local" value="true" />
      <param name="downsample_resolution" value="0.1" />
    </node>

    
    <node pkg="nodelet" type="nodelet" name="hdl_localization_nodelet" args="load hdl_localization/HdlLocalizationNodelet $(arg nodelet_manager)">
      <remap from="/velodyne_points" to="$(arg points_topic)" />
      <remap from="/gpsimu_driver/imu_data" to="$(arg imu_topic)" />
      
      
      <param name="odom_child_frame_id" value="$(arg odom_child_frame_id)" />
      
      
      <param name="use_imu" value="$(arg use_imu)" />
      <param name="invert_acc" value="$(arg invert_imu_acc)" />
      <param name="invert_gyro" value="$(arg invert_imu_gyro)" />
      <param name="cool_time_duration" value="2.0" />
      
      <param name="enable_robot_odometry_prediction" value="$(arg enable_robot_odometry_prediction)" />
      <param name="robot_odom_frame_id" value="$(arg robot_odom_frame_id)" />
      
      
      <param name="reg_method" value="NDT_OMP" />
      
      <param name="ndt_neighbor_search_method" value="DIRECT1" />
      <param name="ndt_neighbor_search_radius" value="2.0" />
      <param name="ndt_resolution" value="1.0" />
      <param name="downsample_resolution" value="0.1" />
      
      
      <param name="specify_init_pose" value="true" />
      <param name="init_pos_x" value="0.0" />
      <param name="init_pos_y" value="0.0" />
      <param name="init_pos_z" value="0.0" />
      <param name="init_ori_w" value="1.0" />
      <param name="init_ori_x" value="0.0" />
      <param name="init_ori_y" value="0.0" />
      <param name="init_ori_z" value="0.0" />

      <param name="use_global_localization" value="$(arg use_global_localization)" />
    </node>

    <node pkg="hdl_localization" type="plot_status.py" name="plot_estimation_errors" if="$(arg plot_estimation_errors)" />

closed 2025-04-30 by emirhancibir

Clushio, 2025-05-05:
I don't have a wheel odometer either. How did you solve it?

emirhancibir, 2025-05-26:
> I don't have a wheel odometer either. How did you solve it?

It should be configured like this:

enable_robot_odometry_prediction must be set to false

odom_child_frame should be set to base or base_link

However, if you're experiencing issues, it's likely due to a broken TF tree.
You need to ensure that your TF tree only includes the map -> base (or map -> base_link) transformation, and the rest of the tree (i.e., base -> ...) should be published by robot_state_publisher.

Please double-check your TF setup.
If you can share the exact error you're getting and your current TF tree, I’d be happy to help you debug it further.